Project-CETI/whale-tag-embedded #12: Add FPGA Verilog and related documentation to the repository

MattCummings0517, 2022-06-30:
The FPGA RTL should be added to the repository and tracked as there are interdependencies with the code running on the Raspberry Pi

MattCummings0517, 2022-07-08:
@PeterMalkin -  Where would be an appropriate location for the Verilog RTL code in the repo?  It's for reference only and won't be compiled at build (it needs to compiled in the Xilinx IDE used to develop the design - the output used by the Tag is the "top.bin" bitstream that we keep in /opt/ceti-tag-data-capture/config.

MattCummings0517, 2022-07-08:
Anyone working on the FPGA-to-Pi interfaces would need to understand this Verilog code...

PeterMalkin, 2022-07-10:
I see two options:

packages/ceti-tag-data-capture/src/rtl,   or
/rtl

The first one is preferable if we could add xilinx toolchain and build the binary from RTL at build time. Its been awhile since I looked at open source toolchains that are not part of IDE, but I *think* that can be done. 
If it is possible to build the binary - the thinking is that you will have everything you need in one place to build and deploy the data capture application in one .deb. The .bin file would be built with the C code and the output put in the config/ folder, then packaged into the .deb file as we do today. 

However, if you never plan to build the FPGA code with the application - like if we cannot find a toolchain to build it outside Xilinx IDE, then I guess we can stick it anywhere. Even in the top level folder. In that case someone would have to manually keep track of what version of the binary gets put into the C application to be packaged. Same as we do today. 

Having all the source code be trackable by git and having a build system that is automated and can produce reliable binaries from source of course is nice. But only if we can find the toolchain, and assume that we will be using the RPi based system for some time (which Michael seemed to have indicated)

MattCummings0517, 2022-07-13:
I am planning to relocate the directories used by the Xilinx IDE for rtl and ipcore into the repository at /src/fpga/rtl  and /src/fpga/ipcore.  I may also create a directory for FPGA documentation that explains how the FPGA is bridging the Pi to the ADC, how the main FIFO and mux works and other information.  At a later time we can consider integrating the Xilinx build steps if it makes sense.  Overall, this effort is relatively low priority at the moment - but if the v2.1 architecture of Rasp Pi + Xilinx is to be supported this is an important.  As a reminder, the primary role that the FPGA plays in the system is that of a multiplexer and an elastic buffer (i.e. a FIFO) between the ADC and the Raspberry Pi.

MattCummings0517, 2022-07-14:
RTL and ipCore files committed.  Closing the issue.

closed 2022-07-14 by MattCummings0517