Project-CETI/whale-tag-embedded #7: Raspberry Pi Serial Port Configuration

MattCummings0517, 2022-06-30:
The v2.1 Tag design uses the Pi's serial port to communicate with the Recovery board.  The Pi must therefore be configured with the serial port enabled and console login from serial port disabled.

PeterMalkin, 2022-07-08:
I think this is what you want. However, I do not have hardware to test on. So please give it a spin if you can

https://github.com/Project-CETI/whale-tag-embedded/commit/e50d68316715d5c5653c49ccf530c09676240429

MattCummings0517, 2022-07-08:
Will test today along with the posix signal feature and report back.

MattCummings0517, 2022-07-08:
Something broke with the ssh at some point along the way.  The most recent commits (prior to those pushed today) were just data-capture .deb changes so I didn't test everything from the ground up - just installed the data capture .deb on the targets and everything was OK.  However, today I am finding that if I build from these earlier commits the Pi is refusing my attempt to connect.  Not sure why, I haven't seen this behavior before.  The device is present on the network and the app is running, but all ssh attempts return "connection refused" .

I'm going to roll it back a bit further to see if I can find a baseline.  Or maybe I am missing something obvious?

MattCummings0517, 2022-07-08:
Another update - it looks like the issue is that it just takes a few minutes for the network to settle, and then the host key needs to be updated.  So, need to be a little patient on the first boot after making a brand new SD card.

The serial port is enabled and working as planned.  Closing this issue.

closed 2022-07-08 by MattCummings0517