Project-CETI/whale-tag-embedded #8: Stopping the ceti-tag-data-capture service can take up to 5 minutes

MattCummings0517, 2022-06-30:
It would be good to have a faster way to stop the service.  In a deployment use case, the Pi will be shutdown when the battery reaches a critical level.

PeterMalkin, 2022-07-06:
Really the problem is that all threads have their main loops as ```while (1)```, and the source code never expects that loop to exit.
Instead we need to introduce a global static variable to use in the while loops, and properly process posix signals. In case of signals recieved the loops should quit and each thread needs to gracefully terminate.

MattCummings0517, 2022-07-06:
Agreed - note that the command/response named pipe interface uses some form of this idea, but it is only applied to the cmdHdlThread - there is global flag called "g_exit".  If the string "quit" is received on the command pipe, g_exit is set and the program cleans up and exits.  This may be a bit clumsy versus posix signals. Maybe the already exisiting g_exit global can be used to break out of the while loops?
//-----------------------------------------------------------------------------
// Globals
//-----------------------------------------------------------------------------
FILE *g_cmd = NULL;  //these are the named pipes
FILE *g_rsp = NULL;
char g_command[256];
int g_exit = 0;
int g_cmdPend = 0;

// Main Loop  listens to the command pipe
    while (!g_exit) { // main loop runs the hearbeat and handles commands
        usleep(100000);
        if (g_cmdPend) {
            hdlCmd();
            g_cmdPend = 0;
        }
    }

    printf("Canceling Threads\n");
...cancel threads, clean up and exit the whole app.

int hdlCmd(void) {
    char cTemp[256];
    //-----------------------------------------------------------------------------
    // Part 1 - quit or any other special commands here
    if (!strncmp(g_command, "quit", 4)) {
        printf("Quitting\n");
        g_rsp = fopen(RSP, "w");
        fprintf(g_rsp, "Stopping the App\n"); // echo it back
        fclose(g_rsp);
        CETI_LOG("Application Stopping");
        g_exit = 1;
        return 0;
    }

PeterMalkin, 2022-07-07:
Yep. Good point. We just need to use the same variable to be used in all threads, and add posix signal handling.
Either myself, or PeterN will take a look

PeterMalkin, 2022-07-08:
One more question - if you use the command interface and type `quit` - do you still get those 5 minutes of waiting before the whole program shuts down? Or does cancelling threads does it pretty much right away? I wish I could try it myself - just dont have the hardware

PeterMalkin, 2022-07-08:
Give this a spin as well. Should probably work, but needs testing. Specifically what happens with the data in the files when the application is stopped.

MattCummings0517, 2022-07-08:
Thanks Peter, will test today.

MattCummings0517, 2022-07-08:
Works as designed.  Nice!  Thanks...

closed 2022-07-08 by MattCummings0517