Project-CETI/whale-tag-embedded #24: Need to form proper RIFF header and capture WAV files, not raw

PeterMalkin, 2022-08-29:
Today we dump the audio data directly on disk, resulting in .raw files with audio straight from ADC.
However this has been causing problems in that the information of the actual format is not present. Therefore during the flac encoding of the audio after downloading data onto a computer from whale tag, the only way to know the format is to go get the version of the embedded software on the tag and check for the encoding of the files.

To fix this I think we should actually use proper WAV files that do encode this information. As such, we'd need to form a proper RIFF header for capturing the audio data, not just dump the raw samples.

MattCummings0517, 2022-08-30:
This also allows for more flexible use of the data acquisition system.  Currently the tags are only used at 96 kSPS, 16-bits, 3 channels - it is hardcoded.  Attaching the RIFF will allow future expert users to have flexibility with these settings without needing to otherwise informing the consumer of the files.  

Also, as part of this, I suggest creating some new configuration variables that the program can reference:  BIT_DEPTH, SMPL_RATE, NUM_CHAN.  These could go in /opt/config/ceti-config.txt

PeterMalkin, 2022-11-17:
The audio data comes out from FPGA in big endian format. RIFF does not support that unfortunately.
So instead, lets implement FLAC compression for the systems that are based on raspberry pi zero 2, which has enough CPU capacity.

PeterMalkin, 2023-01-31:
Flac compression is in place for v2_2 that targets raspberry pi zero 2. Since it is the only platform that has enough compute to run compressor realtime. The old versions of the tags will still save raw audio data which will be encoded during offload. For data offload there's now a convenience script that can do offloading and encoding automatically, if needed, or just offload flac directly (https://github.com/Project-CETI/data-ingest/blob/main/scripts/tag.sh)

closed 2023-01-31 by PeterMalkin