Project-CETI/whale-tag-embedded #26: Micro SD Card Storage Fills Up and Interferes with SSH

MattCummings0517, 2022-10-10:
An issue was discovered when V2.1 Tag no. 3 could not be accessed via SSH wireless or USB Ethernet at all.  Based on the LEDs, the tag otherwise appeared to be working properly.

After leaving the tag running overnight, the main application was found to have stopped on its own for some reason...however, in this condition was able to connect and found the uSD card full (primarily audio data).  This suggests two improvements to avoid lockouts like this:

1) During data acquisition, check disk space periodically and stop saving data before the disk is full (10% spare space?)
2) Provide a short time window after boot (1 min?) for an operator to login before the data-capture service starts.  The capture service is very CPU intensive and bogs down SSH communications.

PeterMalkin, 2022-10-13:
We should also add functionality that would check for the sdcard space available and stop the data capture service if there's no space left to write to.

Ideally at that point we should forcefully remount the system readonly to prevent potential data loss from accidental brown out.

pxno1964, 2022-10-13:
I've been really busy lately with training and getting up to speed on my
new position within Google, in combination with some issues of personal
nature, but I can probably start looking at this next week. I'll sync up
with PeterM early next week.

-PeterN

PeterMalkin, 2022-10-13:
I implemented the delay on start. And theoretically it should not change the way the service is respawned ir the app is crashes. 
https://github.com/Project-CETI/whale-tag-embedded/commit/b80d37ac7626a9386c70beaaf06fc13e5b23c3bd

But the rest of the logic here remains to be done

MattCummings0517, 2022-10-13:
> b80d37a

Thank you Peter. I will build and test over next few days. Matt C.

MattCummings0517, 2022-10-13:
> During the meeting we also discussed that the system fails to shutdown when the battery runs low. I switched to the systemd way of powering the system off.
> 372e74f

Got it. Will include in my testing next week and report back

MattCummings0517, 2022-11-09:
v2.1-4 Adds features to prevent the storage from reaching 100%.  Audio recording is suspended when the uSD has less than 1 GB available.

closed 2022-11-09 by MattCummings0517