Project-CETI/whale-tag-embedded #38: Separate battery monitoring function from data-capture service

MattCummings0517, 2023-01-19:
Because the data-capture service may be stopped by a user during download, the battery monitoring should be made distinct and be run from its own service.  We have seen cases where the battery was inadvertently run down because the service was stopped during download.

PeterMalkin, 2023-01-31:
Here's a first attempt: https://github.com/Project-CETI/whale-tag-embedded/commit/e9da5e9e1a7b559808b76974dda2a931618bffd0

MattCummings0517, 2023-01-31:
Thanks Peter, I will run testing - may not get to it until next week.

MattCummings0517, 2023-02-08:
It turns out that it is not possible to use the pigpio with two services simultaneously.  So...rather than do it this way (separate services for capture and battery monitor), will revert to one service (capture) and build in a new mode that allows an operator to suspend data acquisition while the tag is being offloaded and/or charged.  The operating will instructions will need to change from "stop the ceti-tag-data-capture-service" to "put the tag into data upload/charging mode".  A new pipe command will be created for this method.  Closing this issue and making a new one for the different approach

PeterMalkin, 2023-02-24:
I rewrote the battery monitor service to not use pigpio. But this needs testing
https://github.com/Project-CETI/whale-tag-embedded/commit/d3d050a25d33b9d4e7f47b8ea7e6c9ef526d9118

MattCummings0517, 2023-03-14:
Merged fix into main 3/13/23

closed 2023-03-14 by MattCummings0517