Project-CETI/whale-tag-embedded #65: Stopping the ceti-tag-data-capture service can take up to a minute

MattCummings0517, 2024-05-03:
This is a minor bug but impacts productivity when working on code.  If the service is running and a systemctl stop is issued, often it takes up to a minute to actually stop.  It is likely this has to do with how the audio thread works and associated priorities that prevent other requests from getting through.

rchick, 2024-08-11:
in the file 

https://github.com/Project-CETI/whale-tag-embedded/blob/main/packages/ceti-tag-data-capture/debian/ceti-tag-data-capture.service

could you try adding the key=value

`TimeoutStopSec=10`

under the [Service] section and see if this helps?

msalinoh, 2024-10-21:
Issue resolved in #69 through proper error handling

closed 2024-10-21 by msalinoh